BCN3D/BCN3D-Cura-Windows #3: Feature Request: Parallel print of same GCode on both extruder. 2x pieces with one print.

mrcg, 2016-01-31:
It would be really excellent feature to print same 3d model on both extruder in parallel.
In theory it should be possible but not sure if Sigma FW does support this.

The printing volume needs to be halved from 210x297mm to 105x297.
The benefit of this "duplication" feature is to print 2 pieces in one print.
This will save lot of time !!!

Zorgonaute84, 2016-02-03:
+1 but I don't know if it's technicly possible cause of the sequential traitement of the informations include in the Gcode.

4truemodels, 2016-02-03:
+1 although as @Zorgonaute84 I'm not sure that would be possible because of the hardware specifications of the SIGMA. Having two extruders travelling individually is awesome for dual-printing, but in order to be able for multi-printing is really a trouble.
You can check other machines like staker3D ones for multi-printing. The hardware differences are loud and clear.