BCN3D/BCN3D-Cura-Windows #10: Feature request: Choose main extruder

boletbo, 2016-03-03:
BCN3d Sigma has dual extruders.
Now, on full settings, you can choose which extruder you use to create support but you cannot choose the extruder you use to print the object (or I haven't seen where).
It would be great if you could choose which extruder to use (first or second extruder) as main extruder.
Thank you for all your work!

kitusmark, 2016-03-03:
Hi @boletbo. We're working on that. For doing the workarround, you can edit the g-code file with the notepad and add a T0 or T1 on the first line depending on with which extruder you want to print. 

Let me know if this works for you!

boletbo, 2016-03-04:
Thank you,
This works OK if you print using a single extruder.
If you print using two extruders you must change all T0 to T1 and all T1 to T0. I haven't tried it because I've seen 'T0' or 'T1' strings in 'CURA_PROFILE_STRING' and I don't know if this could affect the printing.

Wim1975, 2016-05-09:
I had the same problem : I thought it was possible (using the quick print settings) to choose which extruder to use. I unchecked all material choices from the first extruder and only checked PLA for the second extruder. Then I made some adjustments in the advanced menu's and when I started printing, I was disappointed to see the first extruder was used. I also noticed both extruders heating up, which is of course not necessary when using only one material.
So I also change the Gcode manually to make sure the right extruder was used.
(the heating of the second extruder I turned of manually after starting a print)

P.S.
Boletbo ... if you want to swap all T0 to T1 and T1 to T0 ... and don't want to mess up ending with only one extruder you can swap the extruders this way (in notepad) :
Replace all 'T0' with 'T9' (or something else not used in the file already, DON'T USE T1 !!!)
Replace all 'T1' with 'T0'
Replace all 'T9' with 'T1'

This should do what you want (I know this is not ideal but at least it works)

Wim1975, 2016-05-22:
I did manage to change the profile so you can choose which extruder to use when only printing in one material (left or right extruder). It was a pain to do so, because S3D is removing codes I added in the scripts to get around this, but finally I found a way. You need to A) make a profile with only ONE extruder in the list, B) Make sure in the temperature profiles you have both T0 and T1 the same temperature (that of the material you plan to use). C) adapt the starting script of the left and right extruder to your needs (heating up the right one and cooling down the other) + your starting settings for purging, C) add a "Tool change" script for left and right where you explicitely turn off the other extruder (S3D will turn it on automatically no matter what you do, but this turns it off again immediately after it). The only thing I need to do now is : Choose my Material, Quality and "Configure for Left (or Right) Extruder Only", AND change the "extruders tool index to Tool 0 for Left or 1 for Right, and I can start printing knowing only the right extruder will be heated and used.