BCN3D/BCN3D-Cura-Windows #9: Check for firmware updates improvements

kitusmark, 2016-03-03:
Hi @remelbourne 

I've noticed a couple of things in the *"check for firmware updates"* menu. 

+ It only works once. When you complete the process and you try to go again to the menu option it doesn't do anything.
+ In the last step of the process **"Finish"** button doesn't do anything too. It should close the window normally.

remelbourne, 2016-03-03:
> Assigned #9 <https://github.com/BCN3D/BCN3D-Cura-Windows/issues/9> to

Thank you Marc! I'll take a look at it

-- 

Rebeca Melbourne
Àrea de R+D+i
<http://www.reprapbcn.com/> <http://www.facebook.com/reprapbcn/>
<http://www.twitter.com/reprapbcn> <http://www.youtube.com/user/reprapbcn>

Nova Edició 2015 del Programa de Màster i Postgrau. Vine a les sessions
informatives:
Disseny i fabricació de Producte
<http://www.fundaciocim.org/formacio/mipg/master-en-enginyeria-i-produccio-integrades-ordinador>
- Automatització <http://www.fundaciocim.org/ca/formacio/master-pair>
- Direcció
de la Producció
<http://www.fundaciocim.org/content/master-en-direccio-de-la-produccio>

remelbourne, 2016-07-11:
Fixed!

closed 2016-07-11 by remelbourne