BCN3D/BCN3D-Cura-Windows #17: Sigma - firmware updating / connection problems

S3D-r, 2016-06-07:
I was not able to connect Sigma with "Cura for BNC3D":

1. Connecting starts...
[image]

2. It seems that there is a connection!
[image]

3. But no, when I try to update the Firmware I get this:
[image]

4. The error log is:
[image]

remelbourne, 2016-07-11:
@S3D-r This should have been fixed in the new version. Let me know if you still have the same problem!

S3D-r, 2016-08-22:
Well, no matter how good "the new version" is, as I'm not able to upload the new version. Actually, that was my primary problem (and still is...).

sigmatester, 2016-08-25:
I have the same problem. Is there any other way to upload firmware?

remelbourne, 2016-09-01:
@S3D-r I meant the new Cura-BCN3D version. We redesigned the firmware updater and it should be working fine. For now it is only a beta version so please let us know if anything is still failing to work. You can find it on the following link: https://github.com/BCN3D/BCN3D-Cura-Windows/releases

@sigmatester You can upload firmware manually by first downloading the source files from this link https://github.com/BCN3D/BCN3DSigma-Firmware/releases and later from Cura-BCN3D go to "Machine -> Install custom firmware" and look for a ".hex" file in the source files you just downloaded which should be under the folder "Compiled Firmware". I recommend you download the latest beta version of Cura-BCN3D since the manual firmware install did not work too well in previous versions.

Hope this helps! Do not hesitate to post any further questions.

S3D-r, 2016-09-01:
Hi, remelbourne!  What does the bug-label mean? Does it mean that you propose a remedy to a problem that is based on some bug in s/w?  

I will give the beta-Cura a change... I'll report, how it went...

remelbourne, 2016-09-01:
Hi @S3D-r, the "bug" label is just a way for us to keep track of issues. What happened to you in the first place was a bug in the code and so this helps us and users see if it already happened or it is something new.

S3D-r, 2016-09-03:
I finally managed to update the firmware.  I had had problems with manual update (of f/w), the updating screen appeared but never progressed:

[image]

The F/W file was not correct. The file v01-1.1.6.hex that I downloaded on 26th July, was 2349kBytes. Yesterday I checked Github and that file was 446kBytes. With that smaller hex-file I managed with f/w updating and the Sigma seems to work now. What a big relief!

remelbourne, 2016-09-05:
@S3D-r  I am glad to hear it is now working.

closed 2016-09-05 by remelbourne