BCN3D/BCN3D-Cura-Windows #19: Unknown travel Issues

Lostinthe907, 2016-06-30:
Hello, I'm am having an unknown Issue I am hoping someone can help me fix. 
this started with my first model, So I updated firmware and tried again. still the same issue is happening. 
While Printing In the Middle of a layer, it will travel to the left or to the front of the printer. leaving a small line of filament to extrude and attach back to the print. I have attached a video to help show what's going on, Because I cannot explain it well enough. 
Not sure if is caused buy having to high of a Min layer time or some other issue. 

any help would be really grateful. 
I am Running the PLA profile for IDEX std. 

-Levi
[Video.zip](https://github.com/BCN3D/BCN3D-Cura-Windows/files/340628/Video.zip)

Lostinthe907, 2016-06-30:
Okay I have Uploaded the Gcode file for help. the issue happens 2 times in just printing the Skirt. 
Maybe someone who understands the gCode better then I do can see the issue their. 
[twoColor1_2_7_10_53new.txt](https://github.com/BCN3D/BCN3D-Cura-Windows/files/340645/twoColor1_2_7_10_53new.txt)

remelbourne, 2016-07-11:
[https://github.com/BCN3D/BCN3DSigma-Firmware/issues/79](url)

closed 2016-07-11 by remelbourne