BCN3D/BCN3D-Cura-Windows #20: Select extruder in start.gcode

omersa, 2016-07-17:
If you add a line T1 in start.gcode to select extruder 2 it is not working right, because Cura export gcode sets extruders and bed temperatures before start.gcode! So it heats extruder 1 instead of 2. I think that start.gcode should always be the first code in final gcode.

Example:
M140 S52.000000
M109 S210.000000
M190 S52.000000
;Sliced at: Sun 17-07-2016 11:16:55
;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 15
;Print time: 1 hour 11 minutes
;Filament used: 3.927m 31.0g
;Filament cost: 1.55 / 7.85
;M190 S52 ;Uncomment to add your own bed temperature line
;M109 S210 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
T1

remelbourne, 2016-07-18:
Thank you for your comment @omersa! Instead of setting T1 after start.gcode you could do either of the two following things:

A)  Write T1 right after the extruder and bed temperatures as follows
M140 S52.000000
M109 S210.000000
M190 S52.000000
#T1#
;Sliced at: Sun 17-07-2016 11:16:55
;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 15
;Print time: 1 hour 11 minutes
;Filament used: 3.927m 31.0g
;Filament cost: 1.55 / 7.85
;M190 S52 ;Uncomment to add your own bed temperature line
;M109 S210 ;Uncomment to add your own temperature line
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops

B) Specify the extruder when setting the temperatures
M140 S52.000000
M109 S210.000000
M190 S52.000000

omersa, 2016-07-18:
Dear @remelbourne, 

I know this can be done, but it is easier if you always print with second extruder that you put the setting in start.gcode as it should be the first position in final gcode. This in my opinion is a bug that the temperatures set by Cura are set before start.gcode. 
If you put T1 after setting of teperatures, then the T0 Will stay at set temperature instead of stay cold.
One can always manually change the final gcode, but then the start.gcode is of no use. Of course, I could do the folowing in start.gcode:
M109 S20.0
T1
M109 S210
But in this case the temperature setting in Cura is of no use. As I print different models with different temperatures, this is not a solution. Solution is that the program is changed in the way, that the temperature settings from Cura are positioned after start.gcode or another file is added (eg. init.gcode), where one could set the primary extruder when printing with only one extruder on a multiple extruder printer.

remelbourne, 2016-07-18:
Dear @omersa, we are currently working on a new simple mode that will let you choose which extruder you wish to print with. I think this is exactly what you are looking for.

I will keep you updated when this new version comes out. Thank you for your comments!